robomechanics/quad-sdk #382: Complex system dynamics with a custom robot

Kate88L, 2023-06-22:
As far as I understand, when updating the matlab script _main_, one can modify the parameters in the dynamics model for a custom robot and generate new casadi functions. However, this matlab script generates simple model only. Is there currently I "simple" way to generate  casadi functions for the complex dynamics model in the existing matlab script? What have you used to generate the necessary casadi functions for complex dynamics model for the spirit robot?

ardalantj, 2023-06-24:
Hi, thanks for using Quad-SDK. I am not sure if I understand what you mean by simple vs complex models here. I am assuming you are referring to centroidal model vs full body model. Currently, we compute the desired ground reaction forces using the centroidal model and pass it to an inverse dynamics controller to realize those as joint torques. In theory, you could customize the existing framework and use the full-body model instead to directly compute joint torques. However, that would require bypassing some of the existing functionality in the robot driver. 

In casadi, we have defined the Lagrangian dynamics of our model and it automatically generates c-code that the NMPC uses from the gen folder, so you could do the same for a different dynamics model.

Kate88L, 2023-07-03:
Thank you for a clear answer. However, when you implemented adaptive and mixed complexity for the Spirit robot, you in fact allowed for a full body model in NMPC. I assume you then generated casadi functions and c-code for the full body model. Did you also use matlab for this code generation? Is there a tool you would recommend to define the dynamic model for a full body dynamics (with urdf support, probably)?

Czworldy, 2023-07-14:
i have the same question @ardalantj

ardalantj, 2023-07-21:
I will close this issue, but feel free to open a new one for any follow ups.

closed 2023-07-21 by ardalantj