robomechanics/quad-sdk #383: Problems of a local planner on a real robot Unitree Go1

mfclabber, 2023-06-23:
Hello. I had a problem with the local planner on a real robot .. . When I run my
"remote_driver.launch" and "roslaunch quad_utils quad_plan.launch reference:=twist logging:=true" 
I get this situation. The robot is standing still, but its trajectory is moving up. 
@jcnorby , How can I fix this?
[image]
[image]
[image]

closed 2023-06-29 by mfclabber

elpimous, 2023-07-15:
@mfclabber how did you solve your problem?

mfclabber, 2023-07-18:
@elpimous The problem was in the robot's IMU, namely in the wrong coordinates relative to the robot. It was also necessary to adapt NMPC for Unitree Go1. These changes can be seen in my[ repository](https://github.com/mfclabber/quad-sdk_go1))

elpimous, 2023-07-18:
Thanks for answer.