robomechanics/quad-sdk #385: How to run  matlab scripts correctly in the quad_logger

nanbwrn, 2023-07-05:
What steps are required to run scripts in quad logs in Matlab?
 I added these two variables according to the steps in the Wiki.
 I added 
trialName='quad_log ';%(The file name of the bag)
namespace='robot_ 1 '; 
The values of these two variables encountered an error while running proscessLog. m
[image]
[image]
What are the detailed steps for running MATLAB scripts? Thank you

nanbwrn, 2023-07-06:
Hi @ologandavid ,

Thank you for your prompt response.
I followed the steps you mentioned and did it again.
The general steps I ran are as follows:
I ran it in MATLAB on a Windows system. I copied the "quad_logger/scripts" folder from the "quad_sdk" to MATLAB in my Windows system, 
and I also copied the "robot_1_quad_log_.bag" file(decompressed file) into it .

After running the program, I encountered the following error:
How can this be resolved?

[image]

ologandavid, 2023-07-06:
Hi @nanbwrn,
If you're looking at a bag file you recorded in sim feel free to comment those lines out given that your state estimate topic  should be empty. In simulation, your state comes directly from gazebo. 

To debug, I would first check to make sure that stateGroundTruth is populated, (comment everything else out) just to make sure matlab is recognizing your desired bag. Did the UI work for you when selecting a bag?

Best,
David

nanbwrn, 2023-07-06:
thank you for you response.
 I'll continue to give it a try.

closed 2023-07-07 by nanbwrn