robomechanics/quad-sdk #388: Running in the step_20cm and rough_25cm environments resulted in an exception

nanbwrn, 2023-07-14:
I replaced the robot with another one (go1) and ran it in the SDK. And I use MATLAB to generate C code for NMPC from the gen folder.
When running global planning in these two environments, I noticed that the robot lost balance in its motion. 
When using the twist command for control, the robot can move normally, but the trajectory of the robot's center of mass appears strange compared to the spirit robot.
The green line representing the center of mass trajectory of go1 doesn't seem to fit well like the spirit robot（the position and length of the green line.）. 

It can be seen that the generated trajectory seems to be located at the chin of go1. Could it be that the error in the body trajectory has led to the failure of the final simulation?

[image]

[image]

[image]

Below are the videos of these two scenarios.

https://github.com/robomechanics/quad-sdk/assets/132453617/9aba810c-05d1-4657-bf94-6d8259aecb0e

https://github.com/robomechanics/quad-sdk/assets/132453617/f6c70cf1-6436-45f8-86ad-220414fa31ee

Czworldy, 2023-07-14:
Which branch you use by the way?

nanbwrn, 2023-07-14:
My system is Ubuntu 18.04, and I am using the main branch.

Czworldy, 2023-07-14:
maybe you should check `desired_height` in `local_planner.yaml`

nanbwrn, 2023-07-15:
Thank you for your response.

nanbwrn, 2023-07-15:
I set the desired_height value to 0.5, which was previously 0.27, but the result was still that the foot touched above the step and fell down.Even when I decrease its value, it still falls down.
I suspect that the trajectory of the center of mass is incorrect, but I'm not sure how to modify it, or there may be other reasons causing this issue.

nanbwrn, 2023-07-15:
It is also strange that it cannot stand properly in the rough_25cm environment.

nanbwrn, 2023-07-19:
I adjusted the robot's initial position according to the steps you mentioned, and now it can stand in the rough_25cm environment and operate properly

nanbwrn, 2023-07-19:
I adjusted the values of swing_kp and swing_kd in quad_utils/config/<robot_name>.yaml under the robot_driver section. Sometimes the robot can jump over obstacles successfully, but other times it falls down.
This is related to the pre-planned path. I will continue to try .
thank you for your help