robomechanics/quad-sdk #390: For different Robot, what params should i watch out?

Czworldy, 2023-07-19:
My question is that if i load a **large mass robot** (more than 50kg) which the size and the mass of this robot is totally different from the `a1` and `spirit` what params should i watch out?

And what i did has been written in my late issue #387 , here i try to load a large size quadruped robot, i can let it stand up but as long as i start the local mpc planner i.e. `reference=twist` it will get very strange plan, before i send the control velocity cmd.

ardalantj, 2023-07-19:
@Czworldy You would need to re-generate the dynamics with the updated mass and inertia properties. This is done in Matlab using Casadi inside nmpc_controller/scripts folder. By updating those parameters and running the script, casadi will automatically generate some files that the NMPC controller uses for the optimization. You have to make sure that you are using the updated dynamics and not the default.

ardalantj, 2023-07-27:
These are the main ones, you may also want to play with kp, kd gains for the joints and standing height which can be different depending on your robot.

closed 2023-07-27 by ardalantj

nanbwrn, 2024-01-07:
hi,@Czworldy

I loaded the Unitree B1 robot in Gazebo, and I made modifications based on the code for controlling the A1. However, when I run the command to make the robot stand, the B1 robot only stands halfway. The size of Unitree B1 is much larger than that of A1 or Go1. I would like to ask how you achieve standing for larger-sized robots. Is the incomplete standing situation due to the larger size of B1?

After running the stand command, it looks like this.

[image]

nanbwrn, 2024-01-07:
hi,@Czworldy 

I loaded the Unitree B1 robot in Gazebo, and I made modifications based on the code for controlling the A1. However, when I run the command to make the robot stand, the B1 robot only stands halfway. Then the robot remained stationary.The size of Unitree B1 is much larger than that of A1 or Go1. I would like to ask how you achieve standing for larger-sized robots. Is the incomplete standing situation due to the larger size of B1?

After running the stand command, it looks like this.
[image]
[image]