robomechanics/quad-sdk #391: Loading Custom Terrain Map

nanbwrn, 2023-07-20:
When I followed the steps in this wiki to create my own map model, an odd situation occurred.
I encountered this situation before adjusting the coordinate system. 
[image]

After adjusting the coordinate system, the robot falls down right after opening the simulation.

https://github.com/robomechanics/quad-sdk/assets/132453617/81df0237-f24e-470a-acdb-cf5cab307db6

https://github.com/robomechanics/quad-sdk/assets/132453617/71f407f2-d2bc-4742-92e3-5d8a07f505b6

nanbwrn, 2023-07-21:
I solved this problem by adjusting the origin of the  terrain map.

closed 2023-07-21 by nanbwrn

ruihuang1124, 2024-04-23:
> I solved this problem by adjusting the origin of the terrain map.

Hi friend, @nanbwrn , I encounter a similar problem, can you briefly describe how you adjust the origin of the terrain map and finally solve this problem?