robomechanics/quad-sdk #395: No directory nmpc_controller/scripts/utils/casadi file

duyongquan, 2023-08-21:
How does the NMPC model find the corresponding formula and code？

ologandavid, 2023-10-06:
Hi @duyongquan, 
The NMPC model grabs the corresponding formula and code it needs from https://github.com/robomechanics/quad-sdk/tree/main/nmpc_controller/.

This is outlined in more detail in the wiki here: https://github.com/robomechanics/quad-sdk/wiki/Tutorial:-Adding-a-New-Type-of-Robot-to-Quad-SDK. The instructions are for adding a new robot, but generating the NMPC scripts and defining its references are outlined here. 

If things are still unclear, follow up with me and I can explain in greater detail.

Best, 
David Ologan

ardalantj, 2023-10-13:
@duyongquan Following up on this. If there is no further issue, we will close this issue.

ologandavid, 2023-10-30:
Closing due to inactivity.

closed 2023-10-30 by ologandavid