robomechanics/quad-sdk #397: Error when displaying multiple robots at the same time

nanbwrn, 2023-09-06:
I'm planning to display multiple robots at the same time in gazebo, however, when I set multiple_robots to true, and then ran gazebo.launch I found the following error:

Has anyone encountered a similar situation before?

[image]

https://github.com/robomechanics/quad-sdk/assets/132453617/5502d62e-0d93-4002-8ca6-5064752d14fc

jcnorby, 2023-09-17:
It looks like whatever is publishing /camera_XXX is using the global namespace rather than relative. If there are two instances of the robot then they need to correctly use [relative names](http://wiki.ros.org/Names) so that the topics/services are /robot_1/camera_XXX and /robot_2/camera_XXX.

nanbwrn, 2023-09-19:
Thank you for your reply. I will follow your suggestion to change the topic names in both the URDF and SDF files. I hope this will yield positive results. Thanks again.

nanbwrn, 2023-09-26:
After addressing the topic publishing issue, upon attempting to visualize two robots again, it was observed that only one robot is displayed correctly, and the following error appears in the terminal.

[ WARN] [1695715402.064407572, 1.000000000]: "robot_2_ground_truth/toe2" passed to lookupTransform argument source_frame does not exist. 

Does anyone have a solution to this issue? Thank you for your responses.

Below is the list of topics being published to.
[multi_robot_rostopic list.txt]：
https://github.com/robomechanics/quad-sdk/files/12725438/multi_robot_rostopic.list.txt

Here are the tf frames below.
[frames.pdf]：
https://github.com/robomechanics/quad-sdk/files/12725516/frames.pdf

Below are the running videos.

https://github.com/robomechanics/quad-sdk/assets/132453617/fc60716d-93d4-4aaf-ab63-d74e468a603f

nanbwrn, 2023-09-26:
[image]

ologandavid, 2023-10-06:
Hi @nanbwrn, 
Just wondering if you had resolved this issue, since I saw in a more recent issue you rendered both robots. If so, could you share the fix, and I can go ahead and close the issue. 

Best, 
David Ologan

nanbwrn, 2023-10-06:
This issue has not been resolved yet. The two robots that I successfully displayed in another issue were  'a1' robots . The new robot I added has not resolved this issue for now.

closed 2023-10-13 by nanbwrn

nanbwrn, 2023-10-13:
I set 'multiple_robots' to 'true' in 'quad_visualization.launch,' and it's not throwing errors for now. I will now close this issue.