robomechanics/quad-sdk #398: Why does the model appear as a white box?

huaizhixu, 2023-09-06:
Hi,

catkin build is OK,when i run <kbd>roslaunch quad_utils quad_gazebo.launch</kbd>,the rviz show a white box.
OS:ubuntu20.04
git branch: Noetic-devel
terminal output:
```
$ roslaunch quad_utils quad_gazebo.launch          
... logging to /home/huaizhi/.ros/log/79581e64-4cc9-11ee-8efd-f92b0d9816de/roslaunch-NUC12WSKi7-319185.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "joy_control_Twip_api" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
started roslaunch server http://NUC12WSKi7:33705/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /global_body_planner/backup_ratio: 0.5
 * /global_body_planner/dt: 0.03
 * /global_body_planner/dz0_max: 2.0
 * /global_body_planner/dz0_min: 1.0
 * /global_body_planner/g: 9.81
 * /global_body_planner/goal_state: [5.0, 0.0]
 * /global_body_planner/max_planning_time: 0.2
 * /global_body_planner/mu: 0.25
 * /global_body_planner/num_calls: 1
 * /global_body_planner/num_leap_samples: 10
 * /global_body_planner/pos_error_threshold: 25
 * /global_body_planner/replanning: True
 * /global_body_planner/startup_delay: 2.0
 * /global_body_planner/t_s_max: 0.25
 * /global_body_planner/t_s_min: 0.12
 * /global_body_planner/trapped_buffer_factor: 7
 * /global_body_planner/traversability_threshold: 0.3
 * /global_body_planner/update_rate: 20
 * /grid_map_filter_node/grid_map_filters: [{'name': 'duplic...
 * /grid_map_filter_node/input_topic: /terrain_map_raw
 * /grid_map_filter_node/output_topic: /terrain_map
 * /grid_map_visualization/grid_map_topic: /terrain_map
 * /grid_map_visualization/grid_map_visualizations: [{'name': 'z_poin...
 * /local_footstep_planner/duty_cycles: [0.5, 0.5, 0.5, 0.5]
 * /local_footstep_planner/foothold_obj_threshold: 0.6
 * /local_footstep_planner/grf_weight: 0.5
 * /local_footstep_planner/ground_clearance: 0.07
 * /local_footstep_planner/obj_fun_layer: traversability
 * /local_footstep_planner/period: 0.36
 * /local_footstep_planner/phase_offsets: [0.0, 0.5, 0.5, 0.0]
 * /local_footstep_planner/standing_error_threshold: 0.03
 * /local_planner/cmd_vel_filter_const: 0.1
 * /local_planner/cmd_vel_scale: 1.0
 * /local_planner/desired_height: 0.27
 * /local_planner/horizon_length: 26
 * /local_planner/last_cmd_vel_msg_time_max: 2.0
 * /local_planner/stand_cmd_vel_threshold: 0.1
 * /local_planner/stand_pos_error_threshold: 0.05
 * /local_planner/stand_vel_threshold: 0.1
 * /local_planner/timestep: 0.03
 * /local_planner/toe_radius: 0.02
 * /local_planner/update_rate: 333
 * /map_frame: map
 * /mesh_to_grid_map_node/frame_id_mesh_loaded: map
 * /mesh_to_grid_map_node/grid_map_resolution: 0.05
 * /mesh_to_grid_map_node/latch_grid_map_pub: True
 * /mesh_to_grid_map_node/layer_name: z
 * /mesh_to_grid_map_node/verbose: True
 * /mesh_to_grid_map_node/world: flat
[... 800 lines cut ...]
Released under the Apache 2 License.
http://gazebosim.org

[INFO] [1694013983.583995, 0.000000]: Waiting for /clock to be available...
[INFO] [1694013983.632106, 0.000000]: Loading model XML from ros parameter robot_description_sdf
[ INFO] [1694013983.632299407]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1694013983.633402999]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[INFO] [1694013983.641626, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.13.17
[ INFO] [1694013983.656673690]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1694013983.658120508]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.13.17
[ INFO] [1694013983.742930439]: Number of polygons: 12
[ INFO] [1694013983.743722314]: Created map with size 9.000000 x 4.000000 m (180 x 80 cells).
[ INFO] [1694013983.743974051]: Published a grid map message.
[ INFO] [1694013983.744052947]: Loaded the mesh from file: /home/huaizhi/yuquan/code/catkin_ws/src/quad-sdk/quad_simulator/gazebo_scripts/worlds/flat/flat.ply. Its frame_id is set to 'map'
[Wrn] [ModelDatabase.cc:340] Getting models from[http://models.gazebosim.org/]. This may take a few seconds.

```
----
[image]

huaizhixu, 2023-09-06:
Some time later, output:
```
[ INFO] [1694013983.743974051]: Published a grid map message.
[ INFO] [1694013983.744052947]: Loaded the mesh from file: /home/huaizhi/yuquan/code/catkin_ws/src/quad-sdk/quad_simulator/gazebo_scripts/worlds/flat/flat.ply. Its frame_id is set to 'map'
[Wrn] [ModelDatabase.cc:340] Getting models from[http://models.gazebosim.org/]. This may take a few seconds.
[Err] [ModelDatabase.cc:394] Unable to get model name[http://models.gazebosim.org/mud_box]
```
---
[image]

huaizhixu, 2023-09-06:
Some time later, output:
```
[ INFO] [1694014994.608271184, 0.377000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_1/robot_description] on the ROS param server.
[ INFO] [1694014994.724330614, 0.377000000]: Loaded gazebo_ros_control.
[Msg] Waiting for model database update to complete...
[INFO] [1694014994.894702, 0.377000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1694014994.898165, 0.377000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1694014994.900287, 0.377000]: Loading controller: joint_controller
[ INFO] [1694014995.000943529, 0.377000000]: Ground Truth State Estimator: <updateRateHZ> set to: 500
[INFO] [1694014995.010302, 0.382000]: Loading controller: joint_state_controller
[INFO] [1694014995.018302, 0.390000]: Controller Spawner: Loaded controllers: joint_controller, joint_state_controller
[INFO] [1694014995.022346, 0.394000]: Started controllers: joint_controller, joint_state_controller
[robot_1/spawn_sdf_model-7] process has finished cleanly
log file: /home/huaizhi/.ros/log/ddeddd86-4cca-11ee-8efd-f92b0d9816de/robot_1-spawn_sdf_model-7*.log

```

pranavgo, 2023-09-09:
Did you find solution to this? I am facing the exact same issue.

huaizhixu, 2023-09-12:
@pranavgo 
 i've solved the problem. The key is to add the following lines to the environment variables
```
#gazebo
export GAZEBO_MODEL_PATH=:/home/huaizhi/yuquan/code/catkin_ws/src/quad-sdk/quad_simulator/spirit_description:/home/huaizhi/yuquan/code/catkin_ws/src/quad-sdk/quad_simulator/other/sensor_description:/home/huaizhi/yuquan/code/catkin_ws/src/quad-sdk/quad_simulator/other/objects_description:/home/huaizhi/yuquan/code/catkin_ws/src/quad-sdk/quad_simulator/gazebo_scripts/worlds
```
I referred to someone's [issue](https://github.com/robomechanics/quad-sdk/issues/319),he suits me

closed 2023-09-12 by ologandavid

NimanthaAdikaram, 2025-07-07:
The issue is due to Gazebo. Version 9.0.0, which supports Ubuntu 18.04, needs to be installed. Update quad_simulator
/setup_deps.sh line `sudo apt install -y gazebo9 -y` with

`sudo apt install -y gazebo9 libignition-fuel-tools1-dev`