robomechanics/quad-sdk #402: issues on noetic_devel_ekf_clean

elpimous, 2023-09-17:
Hello the team : 

1/ when using gazebo + rviz on `roslaunch quad_utils quad_gazebo.launch`,
spirit_description, a1_description and ylo2_description added to ~/.bashrc
Spirit and A1 play correctly on rviz+gazebo
Ylo2 plays ok on rviz, but not under gazebo. Do I miss anything ?

spirit:
[image]

a1:
[image]

ylo2:
[image]

2/under rviz, foot contacts are not same position on Ylo2 : Where should i investigate ?

spirit:
[image]

a1:
[image]

ylo2:
[image]

elpimous, 2023-09-17:
3/ should I need to change this too ? 
https://github.com/robomechanics/quad-sdk/blob/9973ff94a6b28e74fd69b6b51f561b2515f9c59d/nmpc_controller/include/nmpc_controller/quad_nlp.h#L280-L289
not written on tuto : Adding-a-New-Type-of-Robot-to-Quad-SDK

ologandavid, 2023-10-30:
Hi @elpimous, 
Any updates? Quad-SDK's default estimator is still the comp filter. You can specify EKF by either directly changing it in the launch file, or feeding it in as an argument during launch.

Best, 
David Ologan

elpimous, 2023-10-30:
Hello @ologandavid
Happy to here from you
Could you tell me what is the correct branch to use, for a Noetic real robot test, with ekf filtrer ?!

Noetic devel ekf clean didn't work a mont ago, on real (worked on sim)
Take care of you my friend
Vincent

ologandavid, 2023-10-30:
Hi @elpimous, 
Did it at least work for your robot in sim? There are a couple of things you would have to change to get it working on a real robot. First, you would need to input your robot's start configuration, (standing) to initialize the filter, and some tuning is probably required to see the desired results. You should also stop publishing the imu data from rviz, and instead use your robot's imu. I can push some more changes to that branch to make this process a little easier.

Best, 
David Ologan

elpimous, 2023-10-30:
Yeah.
It worked nicely on sim with ylo2.

Oh, I anderstand now why it didn't work as expected on real. 

Perhaps could you push a readme_for_real_usage.md ?!

Sure, modifying files could be cool for me, but some of my friend, like me, need to anderstand and test.. (and our brain isn't as accurate as your lol).
I'm really happy to see you taking care of your users..

So, the story continues.... Lol

ologandavid, 2023-10-30:
Hi @elpimous, 
Yeah, I can make some changes to the setup to help streamline it and add a readme in estimators. 

Best, 
David Ologan

elpimous, 2023-11-01:
@ologandavid 
Please, perhaps could you informed us, here, when you'll make changes on devel ekf clean.
we'll test and report our progresse here, if you want.
We will be patient, for sure.

elpimous, 2023-11-07:
@ologandavid  Hi,
noetic_devel_ekf_clean, on SIMULATION working on ylo2

elpimous, 2023-11-07:
**Questions** : 
1/ `you need to input your robot's start configuration, (standing) to initialize the filter`
- Actually I changed `sit_joint_angles` `stand_joint_angles` `sit_kp` `sit_kd` from quad_util/config/ylo2.yaml
Is it what you were saying ?!

2/`You should also stop publishing the imu data from rviz, and instead use your robot's imu`
- When using `is_hardware=true` in quad_spawn.launch,  the controller runs **robot_driver/ylo2_interface**
- the variable `last_imu_msg_` is feeded with my real IMU. (tested with a simple cout, into Ylo2Interface::recv function
It should be enough, correct ?

3/ Anything else to modify ?

THANKS

ologandavid, 2023-11-07:
Hi @elpimous, 
Sorry I haven't quite gotten to this yet. 
1) When I say input the robot's start configuration, I mean directly in robot_driver's update_state function. In sim, I grab the robot's start state for the estimation from gazebo (estimated_state = last_robot_state_msg). For hardware, you need to do the same, but instead you need to initialize the robot on the starting stand state of your robot.

2) Regarding the IMU, I modified Spirit's urdf to publish to the imu topic in sim, but if you're using your already grabbing it from your robot you don't need to worry about this part. 

Best, 
David

elpimous, 2023-11-08:
@ologandavid 
Does it seems correct for you ?

```
bool RobotDriver::updateState(){
  // If Operating on Hardware
  if (is_hardware_) {
    // grab data from hardware
    bool fully_populated = hardware_interface_->recv(
        last_joint_state_msg_, last_imu_msg_, user_rx_data_);

    ros::Time state_timestamp = ros::Time::now();

    // load robot sensor message to state estimator class
    if (fully_populated) {
      
      //TODO : added for ylo2 real
      estimated_state_ = last_robot_state_msg_;
      last_robot_state_msg_.body.twist.angular = last_imu_msg_.angular_velocity;
      last_robot_state_msg_.joints = last_joint_state_msg_;
      last_joint_state_msg_.header.stamp = state_timestamp;
      last_imu_msg_.header.stamp = state_timestamp;

      state_estimator_->loadSensorMsg(last_imu_msg_, last_joint_state_msg_);
    }
    else {
      ROS_WARN_THROTTLE(1, "No imu or joint state (robot) recieved");
    }...
```

elpimous, 2023-11-18:
Hi @ologandavid ... 
Any help ?

elpimous, 2023-11-23:
Hey @ologandavid 
Small up ?

elpimous, 2023-12-25:
@ologandavid 
Hello, did you have Time to modify /devel code, for Real usage ?
What are the needed nodes, for Real ?
Merry Christmas all.
Vincent

ologandavid, 2023-12-26:
Hi Vincent, 
Thanks for all your patience. 
I'm making some changes to the code to make this process a little easier for you. I don't think you should need any additional nodes outside of robot_driver to get estimation working. If you'd like we could schedule a formal meeting to work out some of the issues you've been having. 

Happy Holidays, 
David Ologan

elpimous, 2023-12-27:
Hi, cool. Could you tell me when it will be done.
About meeting, my english on visio? Would be erratic 😁.
I emailed you. Take your Time , and i Hope you'll not forget me. 😅🥳

closed 2024-03-23 by ologandavid