robomechanics/quad-sdk #403: A strange issue that appeared after I changed the robot to GO1.

hrxsd, 2023-09-26:
Sorry for the bother.
After I replaced the robot with GO1 following the tutorial in the Wiki, the local planner doesn't seem to be working correctly (the robot can stand normally and the global planner is fine).
I then switched the bot back to spirit and the same thing happened (as pictured). This is strange because I followed the instructions in the tutorial the entire time.
I would be grateful for an answer from your subject.
[image]
[image]

ologandavid, 2023-10-25:
Hi @hrxsd,
Did you change anything in the code/ what branch are you currently using? I can't tell the exact issue with just the information provided. 

Best, 
David Ologan

ologandavid, 2023-10-30:
Just checking in on the status of this issue, is there any additional information you could provide?

Best, 
David

hrxsd, 2023-10-31:
@ologandavid Thank you for your attention to this issue, I just added go1 to this project step by step following the guidelines in the Wiki about adding new robots and this problem occurred. 
It's worth noting that when I added only the URDF file, the robot was able to run correctly in the simulation (at this point, no dynamics model had been generated using MATLAB scripts, and no source code had been modified). However, when I added the above as per the tutorial in the Wiki, the problem arose.

closed 2025-07-09 by ologandavid