robomechanics/quad-sdk #405: Error when building quad-sdk.

Nicogrant9, 2023-09-30:
Errors     << local_planner:make /home/a/quad_ws/logs/local_planner/build.make.000.log
/usr/bin/ld: warning: libcoinhsl.so.0, needed by /usr/local/lib/libipopt.so.1, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma57ad_'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma86_solve_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma57id_'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma27bd_'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma77_input_reals_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma28ad_'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma27cd_'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma57bd_'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `mc19ad_'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma77_analyse_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma27id_'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma97_factor_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma97_default_control_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma77_solve_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma77_default_control_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma86_factor_solve_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma86_default_control_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma77_input_vars_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma77_factor_solve_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma77_open_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma97_solve_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma97_finalise_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma86_finalise_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma57ed_'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma86_analyse_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma57cd_'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma97_free_akeep_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma27ad_'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma77_finalise_d'
/usr/bin/ld: /usr/local/lib/libipopt.so.1: undefined reference to `ma97_analyse_d'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/local_planner_node.dir/build.make:264：/home/a/quad_ws/devel/.private/local_planner/lib/local_planner/local_planner_node] 错误 1
make[1]: *** [CMakeFiles/Makefile2:334：CMakeFiles/local_planner_node.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
cd /home/a/quad_ws/build/local_planner; catkin build --get-env local_planner | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

elpimous, 2023-10-01:
hello, did you install ipopt ?
did you follow this step : 
```
chmod +x setup.sh && ./setup.sh
cd ~/catkin_ws
catkin build
```

ologandavid, 2023-10-06:
Hi @Nicogrant9,
This is definetly an issue with the way you installed the solver during the setup. Did you make sure to request a copy of the solver, and put it in the correct folder? 

Best, 
David Ologan

closed 2023-10-25 by ologandavid