robomechanics/quad-sdk #409: local planned can not build successfully,see pics below:

Crazywjw, 2023-11-07:
/usr/bin/ld: warning: libcoinhsl.so.0, needed by /usr/local/lib/libipopt.so.1, not found (try using -rpath or -rpath-link)
[image]
[image]

ologandavid, 2023-11-07:
Hi @Crazywjw, 
This could be a combination of issues, I think the first would be to make sure you requested the right version of the solver, I'm using ma27, so I can confirm that that version works. Next, take a look at these commits from this pending [pull request](https://github.com/robomechanics/quad-sdk/pull/408/commits/15af19a209332b4be6ce72b1cba458220fc21e7e). Try updating these lines in the code, and then rerunning the setup.sh script. and then retrying the build. If the issue persists, please follow up here. 

Best, 
David

ologandavid, 2023-12-19:
Hi, I'm following up on this issue, otherwise I'll mark it as closed. 

Best, 
David Ologan

closed 2023-12-24 by ologandavid