robomechanics/quad-sdk #410: NMPC solving fail , not similar to #288

MusubaPy, 2023-12-07:
Hello gentlemen,

I hope this message finds you well. First and foremost, I want to express my gratitude for your excellent work. However, I have encountered an issue that I am unsure how to resolve:

**"NMPC solving fail"**

I was able to address this error by executing "catkin clean" followed by a subsequent "catkin build." However, rebuilding the project twice each time I enter the docker environment is not ideal due to the associated costs. Strangely, the SECOND rebuild of the project seems to be the resolution for this error. I have reviewed previous open issues, but unfortunately, they did not provide a satisfactory solution.

Here's a summary of my actions:

I am building your project in a Docker container with Ubuntu 18.04 and ROS Melodic. Additionally, I possess an academic license for HSL.

1) Cloned your repository:
`git clone https://github.com/robomechanics/quad-sdk.git ~/catkin_ws/src/`

2) Cloned the Ipopt stable/3.13 repository into quad-sdk/external/ipopt/coinbrew/:
`git clone https://github.com/coin-or/Ipopt.git ~/catkin_ws/src/quad-sdk/external/ipopt/coinbrew/`

3) Cloned the ThirdParty-HSL repository and placed it in quad-sdk/external/ipopt/coinbrew/ThirdParty/HSL, including the coinhsl folder:
`git clone git@github.com:coin-or-tools/ThirdParty-HSL.git ~/catkin_ws/src/quad-sdk/external/ipopt/coinbrew/ThirdParty/HSL`

4) Build the project:
```
cd ~/catkin_ws/src/quad-sdk/
./src/$DIR_NAME/setup.sh

cd ./src/$DIR_NAME/external/ipopt/coinbrew/Ipopt/
./configure
make
make install

cd ../ThirdParty/HSL
./configure
make
make install

cd ./../../../../../../../
catkin build
```
5) Additionally, I update and upgrade the system packages:
```
apt update
apt -y upgrade
```
6) Finally, I source the bash files:

```
source ~/.bashrc
source ~/../catkin_ws/devel/setup.bash
```

Despite following these steps, the "NMPC solving fail" error persists even after the initial build. But after performing
```
catkin clean
catkin build
```
the problem is solved. What can I do to build the project **only once**? Your insight to solve this problem would be greatly appreciated.

Thank you for your time and assistance.

Best regards

elpimous, 2023-12-07:
hi friend, first, could you change nmpc_controller.cpp L168 " app_->Options()->SetIntegerValue("print_level", 0);" to value =5
it will verbose ipopt feedback during run.

Launch code and see error !

perhaps could you try to run : chmod +x setup.sh && ./setup.sh  !!! (https://github.com/robomechanics/quad-sdk/wiki/1.-Getting-Started-with-Quad-SDK)

MusubaPy, 2023-12-08:
Perhaps the attached files, will help you reproduce the error:

1) It is necessary to put "coinhsl" in the folder "requirements"
2) (optionally) install docker using `./install_docker.sh`
3) `./build_docker.sh -n`
4) `./run_docker.sh -n`

Inside docker:
```
cd catkin_ws/
./build.sh
source ~/.bashrc
source devel/setup.bash
```

[docker.zip](https://github.com/robomechanics/quad-sdk/files/13610792/docker.zip)

MusubaPy, 2023-12-18:
Hello, gentlemen.

In the end, I managed to solve the issue with the need for two builds of the project. This solution worked for me: https://github.com/casadi/casadi/issues/2769

In short
> What I ended up doing was compiling the HSL code using [ThirdParty-HSL](https://github.com/coin-or-tools/ThirdParty-HSL), which is one of the options that Ipopt gives in its [documentation](https://coin-or.github.io/Ipopt/INSTALL.html). The shared library `libcoinshsl.so` gets created in `/usr/local/lib`, but since Ipopt looks for `libhsl.so` make a symbolic link from `libcoinhsl.so` to `libhsl.so`.
> `ln -s /usr/local/lib/libcoinhsl.so /usr/local/lib/libhsl.so`
> Then make sure Ipopt knows where to look.
> `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib`

closed 2023-12-18 by MusubaPy