robomechanics/quad-sdk #412: The UnitreeB1 robot (50kg, 1126467636mm) cannot stand completely when using the stand command.

nanbwrn, 2024-01-17:
hello everyone,

After I successfully loaded the B1 robot in Rviz, I sent the stand command to the robot. However, I found that the robot couldn't stand completely. I followed the steps in the wiki to add the robot and also added the b1.yaml file as shown in the following image:

[image]

[image]

I generated the dynamics C++ code for the B1 robot. When I execute the motion planning command, the result is as shown in the following image. It seems that the issue may be caused by the robot standing normally.

[image]

I've seen people make the robot stand normally by changing the initial origin of the rpy of  knee joint in the URDF. However, when I continuously modify this joint's starting angle, I find that the robot still cannot stand. Do you have any other debugging methods, and how can I solve this issue?

ologandavid, 2024-02-22:
Just checking in, on if you figured out a solution to this? Otherwise I'm gonna close this issue as resolved. 

Best, 
David Ologan

ologandavid, 2024-02-29:
Would you be willing to share me the code that you're working with in sim, I can take a look and see if I can deduce whats causing your issue. 

Best, 
David Ologan

nanbwrn, 2024-03-02:
hi, @ologandavid 
Thank you for your help. As I'm not very familiar with how to upload folders to GitHub, I have sent the code to your Google email in a compressed file. If it's convenient, could you please take a look at the Google email?
My Google email is xiangzhang870@gmail.com.
If you haven't received it, please let me know. Thank you.

ologandavid, 2025-07-09:
Very late, but I think the simulation controller has hard coded torque limits in the controller plugin that might need to be changed. Let me know id that works, otherwise I'm going to mark this as closed.