robomechanics/quad-sdk #413: Show estimated terrain

DonovanZhu, 2024-02-13:
Hi:

I notice an excellent project **Proprioception and Tail Control Enable Extreme Terrain Traversal by Quadruped Robots** based on quad-sdk. In this project the estimated terrain is shown like the following photo:
[image]

Unfortunately, I can't find the open source code for that project. Is this terrain estimation method included in this SDK? Or does it have an API to display real-time estimated terrain?

Thanks!

ologandavid, 2024-02-13:
Hi, @DonovanZhu, 
The code for this project should be in branch add-tail-controller-latest within Quad-SDK. For more specific questions, I would recommend you reach out directly to @yanhaoy, who authored the paper and can probably provide more insight. 

Best, 
David Ologan

DonovanZhu, 2024-02-14:
> Hi, @DonovanZhu, The code for this project should be in branch add-tail-controller-latest within Quad-SDK. For more specific questions, I would recommend you reach out directly to @yanhaoy, who authored the paper and can probably provide more insight.
> 
> Best, David Ologan

Thank you so much for your reply @ologandavid! It helps a lot.

closed 2024-02-14 by DonovanZhu