robomechanics/quad-sdk #414: Does it support parallel quadruped robots

Ihtbfas, 2024-03-10:


ologandavid, 2024-03-11:
Hi @Ihtbfas, 
Yes, Quad-SDK does support parallel quadruped robots. Instructions on spawning in and running multi-robots can be found in the wiki. Note that at the moment the global planner will plan for each robot individually, without consideration of the other robot's trajectory. I got a multi-robot CBS planner working with Quad-SDK late last year, but haven't gotten around to cleaning it up and formally adding it to the code base. Hopefully I can add that soon!

Best, 
David Ologan

Ihtbfas, 2024-03-14:
Hi @ologandavid ,
May I ask if it is possible to set a new target point on rviz after the robot reaches its default target position?

ologandavid, 2024-03-16:
That should be possible. You can do it two different ways. First, you can publish a point in the rviz interface to set a new goal point for the global planner. (See the image below) Publishing a goal state directly to the topic works too, but I find the former to be easier. 

[image]

Best, 
David

ologandavid, 2024-03-22:
Checking in on the status of this issue, I'll go ahead and close it if resolved.

Ihtbfas, 2024-03-23:
May I know what sensor you used? I didn't see any LiDAR. Did you use  foot sensors?

ologandavid, 2024-03-23:
Currently, to my knoweldge we don't have a LiDAR implemented, and since the interface would be sensor specific, that's something you would have to implement yourself and incorporate into the code base. An older branch someone might have worked on might, so I can ask around and get back to you on that.

The quadruped platform we have in lab doesn't have built in force sensors in the feet. There are branches on the repo that use an observer to do this, but the main and devel branches assume an terrain map input beforehand and expected foot contact timing from the local planner. If you have any implementation questions let me know!

Best, 
David Ologan

Ihtbfas, 2024-03-23:
I understand. Thank you for patiently answering.

closed 2024-03-23 by Ihtbfas