robomechanics/quad-sdk #415: Adapting Quad-SDK on Aliengo for autonomous navigation project

asuzukosi, 2024-03-22:
Hello, We are working on implementing autonomous navigation for a Unitree Aliengo robot in an unstructured environment for data collection. 
We are currently using the RPLidar a2m8 with HectorSLAM, we are using ROS kinetic on the Aliengo Unitree. Is Quad-SDK the best approach to our problem and how can we adapt it to our use case

ologandavid, 2024-03-22:
Hi @asuzukosi,
I'm happy to hear you're using Quad-SDK in your own work. 

I haven't personally tested Quad-SDK in ROS Kinetic, but we have confirmed working version on both Noetic and Melodic and I think ROS is mostly backwards compatible. If you would like to incorporate mapping from your RPLidar a2m8, I think the best approach would be to write that into it's own node, and publish a terrain map to the global planner to use the rest of the planning and control stack. Also see the wiki for instructions on adding a new robot to the code base. https://github.com/robomechanics/quad-sdk/wiki/Tutorial:-Adding-a-New-Type-of-Robot-to-Quad-SDK

Keep us updated on your progress or if you have any additional implementation questions. 

Best, 
David Ologan

asuzukosi, 2024-04-04:
Thank you @ologandavid, I'm having issues when running the planning node. I seems the local planner isn't running. I am using ros noetic. Please how can I resolve this

ologandavid, 2024-04-04:
HI @asuzukosi, 
Could you provide an image of the error message you're facing? Off intuition its probably an issue with the solver or the way your launching the node, but I can't say for certain until I see what you're dealing with. 

Best, 
David Ologan

ologandavid, 2024-04-26:
@asuzukosi , Any updates on this? Otherwise I'll go ahead and close it.

closed 2024-05-06 by ologandavid