robomechanics/quad-sdk #418: Launch Fails for: roslaunch quad_utils quad_plan.launch reference:=twist logging:=true

sanketlokhandegithub, 2024-03-26:
Hi,

I installed everything as per the instructions and tried to launch with 

$ roslaunch quad_utils quad_plan.launch reference:=twist logging:=true

I see the following errors:

Has anyone encountered this issue recently?

process[robot_1/plan_publisher-6]: started with pid [19004]
[ INFO] [1711413936.284909206]: Can't find param robot_type on rosparam server, loading default value.
[ERROR] [1711413936.285291125]: Can't find param /topics/terrain_map from parameter server
[ERROR] [1711413936.285812148]: Can't find param /map_frame from parameter server
[ERROR] [1711413936.288494464]: Can't find param /local_planner/update_rate from parameter server
[ERROR] [1711413936.288588426]: Can't find param /local_planner/timestep from parameter server
[ERROR] [1711413936.288672785]: Can't find param /local_planner/horizon_length from parameter server
[ERROR] [1711413936.288752334]: Can't find param /local_planner/desired_height from parameter server
[ERROR] [1711413936.288830947]: Can't find param /local_planner/toe_radius from parameter server
[ERROR] [1711413936.289355458]: Can't find param /local_planner/cmd_vel_scale from parameter server
[ERROR] [1711413936.289472302]: Can't find param /local_planner/last_cmd_vel_msg_time_max from parameter server
[ERROR] [1711413936.289561104]: Can't find param /local_planner/cmd_vel_filter_const from parameter server
[ERROR] [1711413936.289642294]: Can't find param /local_planner/stand_vel_threshold from parameter server
[ERROR] [1711413936.289721887]: Can't find param /local_planner/stand_cmd_vel_threshold from parameter server
[ERROR] [1711413936.290314712]: Can't find param /local_planner/stand_pos_error_threshold from parameter server
local_planner_node: /usr/include/eigen3/Eigen/src/Core/CwiseNullaryOp.h:74: Eigen::CwiseNullaryOp<NullaryOp, MatrixType>::CwiseNullaryOp(Eigen::Index, Eigen::Index, const NullaryOp&) [with NullaryOp = Eigen::internal::scalar_constant_op<double>; PlainObjectType = Eigen::Matrix<double, -1, -1>; Eigen::Index = long int]: Assertion `rows >= 0 && (RowsAtCompileTime == Dynamic || RowsAtCompileTime == rows) && cols >= 0 && (ColsAtCompileTime == Dynamic || ColsAtCompileTime == cols)' failed.
[robot_1/local_planner-3] process has died [pid 19001, exit code -6, cmd /home/sanket/quadsdk_ws/devel/lib/local_planner/local_planner_node __name:=local_planner __log:=/home/sanket/.ros/log/28756d32-eb0a-11ee-aa78-6805ca3a8f2e/robot_1-local_planner-3.log].
log file: /home/sanket/.ros/log/28756d32-eb0a-11ee-aa78-6805ca3a8f2e/robot_1-local_planner-3*.log

closed 2024-03-26 by sanketlokhandegithub

MusubaPy, 2024-06-24:
I believe you could make the world a much better place by telling us all how you solved your problem, instead of silently closing the thread.

sanketlokhandegithub, 2024-06-25:
> I believe you could make the world a much better place by telling us all how you solved your problem, instead of silently closing the thread.

I closed the issue as I couldn't reproduce the error. The quad-sdk runs just fine if all the instructions are followed as it is. Are you getting the same error?

MusubaPy, 2024-06-25:
I agree with you that, generally, if all instructions are followed, there should be no errors.

The error mentioned above usually occurs due to an incorrect call to load_global_params.launch.

In my case, I encounter the same issue when trying to run robot_driver.cpp with a real Unitree Go2 robot. Currently, I haven't fully figured out why robot_driver.cpp doesn't read information from the topics correctly, which causes it to crash.

ssaigarimella, 2024-08-23:
@MusubaPy I am getting the same error with the Go2 robot in gazebo simulation (not real hardware) with robot_driver.cpp. Were you able to solve this?