robomechanics/quad-sdk #419: LocalPlanner runs too slowly and takes too long time

ruihuang1124, 2024-03-26:
Hi everyone, I'm trying to use quad-sdk in my work, so I forked the repository into my own account as: https://github.com/ruihuang1124/quad-sdk, and the branch of my code is dev/ray. I encountered a local planner that took too long to run issue, I tried different ipopt linear solvers (ma27, ma57), but the results were similar. The problem is as shown in the screenshot below. You can see that LocalPlanner took 10+ms and caused the frequency of the /robot_1/local_plan topic to be too low. (The expectation is 333hz, but the actual release frequency is only 80-100hz as shown in the picture). The configuration of my computer is CPU: AMD Ryzen 7 5800H and GPU: GeForce RTX™ 3060 6GB, so I think the computer's performance is sufficient?
[image]

ologandavid, 2024-03-27:
Hi @ruihuang1124, 
Assuming your branch was branched off of devel, the local planner should be running raster than its currently running on your system. I would also double check your install of the solver, see Issue 288 for more details on this: https://github.com/robomechanics/quad-sdk/issues/288

When you tried different solvers, did you also make sure to change line 208 on nmpc_controller.cpp to the corresponding solver? Also it would help if you could detail what modifications you made to the devel branch, so I could help further. I don't believe you need to worry about your computer configuration at all, its performance should be more than enough. For reference, the local planner should take 2-3 seconds, 

Best, 
David Ologan

ruihuang1124, 2024-03-29:
Hi friend, @ologandavid 

Thank you for your reply, I do change line 208 on nmpc_controller.cpp to the corresponding solver, and the log information shows that I was using the right solver as I change the "print_level" from "0" to "5". Actually, I found that my problem may not be caused by computer configuration or code writing: I tried configuring this repo on another brand-new laptop (the code version is still branch dev/ray in https://github.com/ruihuang1124 /quad-sdk), when running the code after just configuring the software, the local planner took only 2-3ms as you mention, everything looked good. However, when I further installed pinocchio (https://github .com/stack-of-tasks/pinocchio) warehouse and hipim (https://github.com/giaf/hpipm) and their dependencies. At this time, I ran the local planner and found that it took a very long time (larger than 50cm), I realized that maybe the newly installed software had changed some of my quad-sdk dependencies, so I went into my catkin_ws workspace and deleted all the build and devel files and then completely recompiled and installed quad-sdk, then I ran local_planner and found that it only took 2-3ms. So under the current quad-sdk software version: I think we can completely recompile and install the quad-sdk related software when we encounter a situation similar to mine. I hope it can give people encounter similar problems a reference solution.

Best,
Ray.

closed 2024-03-29 by ruihuang1124