robomechanics/quad-sdk #420: Global Body Planner issue

IonaGuyomarch, 2024-03-27:
Hi everybody, 

I have some issue when launching the global body planner: I'm able to publish the clicked point and in the console I have this: 

```shell
Solve time: 0.0148776 s
Vertices generated: 6
Path length: 4.38682 m
Path duration: 8.69032 s

complexity_schedule = 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
``` 

The problem is that the robot doesn't move at all, even if it can calculate the path. How can I solve this?

ologandavid, 2024-03-27:
Hi @IonaGuyomarch,
When the planner stops at that step, it's most likely an issue with the local body planner, not the global body planner. The global body planner is outputting a body trajectory for the robot to follow, but the local planner can't resolve it. This issue could stem from the way you defined your robot, the way you installed the solver, or your hardware setup. I'd need more information to help you narrow down exactly what's going wrong. 

Best, 
David Ologan

IonaGuyomarch, 2024-03-28:
Hi David, thank you for your quick answer! Well I've build my workspace again and found the problems: there were two functions supposed to return a value that weren't returning nothing. 

1. In file global_body_planner/src/global_body_planner.cpp line 182:

```c++
bool GlobalBodyPlanner::callPlanner() 
``` 
I added "return false" line 310. 

2. In file global_body_planner/include/global_body_planner/global_body_plan.h , line 115 I replaced:
```
inline double setPublishedTimestamp(ros::Time timestamp) {
    published_timestamp_ = timestamp;
  }
```  
with:
```
inline ros::Time setPublishedTimestamp(ros::Time timestamp) {
    published_timestamp_ = timestamp;
    return published_timestamp_;
  }
```  

I hope this will help those who are stuck there! 

Kind regards,
Iona.

closed 2024-03-28 by IonaGuyomarch