robomechanics/quad-sdk #422: The map loaded in the simulation has been simplified

nanbwrn, 2024-04-20:
hi,everyone

I created a map in SolidWorks that includes rebar, but when I start the simulation and load the map into rviz, the rebar (which has a 
diameter of 10mm) doesn't show up. It seems like the map might have been simplified. Has anyone else encountered this issue? What parameters do I need to adjust to make the rebar visible?

thank you very much.

[image]

[image]

nanbwrn, 2024-04-23:
ok

nanbwrn, 2024-04-23:
ok,I'm also a newcomer to this field, so let's learn from each other.

ologandavid, 2024-04-24:
Hi @nanbwrn @ruihuang1124,
When you launch visualization, i.e quad_gazebo.launch try adding the flag gui:=true. The rebar you've added might spawn into gazebo, and not rviz, since rviz is used primarily for visualization instead of an actual representation of the environment. 

Best, 
David Ologan

ologandavid, 2024-05-06:
Updates on this before I go ahead and close it as completed?

nanbwrn, 2024-05-13:
I apologize for the late response; I've been held up with some matters recently.

closed 2024-05-13 by nanbwrn