robomechanics/quad-sdk #425: Using latest version of HSL (2024.05.15) breaks setup.sh script

DanielChaseButterfield, 2024-05-16:
I followed the instructions in the [quad-sdk wiki](https://github.com/robomechanics/quad-sdk/wiki/1.-Getting-Started-with-Quad-SDK), using the `devel` branch, and using the most recent version of Coin-HSL (2024.05.15). 

However, doing this leads to a ThirdParty HSL build error in the setup script, which later leads to an ```NPMC solving fail``` when running the simulator.

I was able to solve it by installing an earlier version of CoinHSL (2022.11.09).

ologandavid, 2024-05-18:
Hi @DanielChaseButterfield, 
In the coming weeks I'll review the setup scripts and make sure they reflect the updates to CoinHSL. Thanks for bringing this to our attention!

Best, 
David Ologan

DanielChaseButterfield, 2024-05-22:
I was curious as to whether this issue was related to ThirdParty-HSL itself, and it looks like it might be as their code also fails to build HSL on its own with the newest version. Therefore, this issue might be resolved when they update their code.

Here's the corresponding issue on ThirdParty-HSL for reference (https://github.com/coin-or-tools/ThirdParty-HSL/issues/24).