robomechanics/quad-sdk #427: [robot_1/local_planner-2] process has died (different from #318!)

ryanz2023, 2024-07-06:
# Description
This happened when I published /robot_1/clicked_point in order to publish the goal. The global planner had successful planned with no errors, the local planner process had died (maybe because the assertion failed in eigen3 library). The log is shown below.
# Log
```
[ INFO] [1720259754.210421905, 628.775000000]: In reset mode
[ INFO] [1720259754.212852163, 628.777000000]: valid and shorter or previous plan not valid
Solve time: 0.00214676 s
Vertices generated: 6
Path length: 4.99255 m
Path duration: 9.98333 s

[ INFO] [1720259754.248552296, 628.802000000]: LocalPlanner took 10.467 ms
[ INFO] [1720259754.283325307, 628.827000000]: valid and shorter or previous plan not valid
Solve time: 0.00243434 s
Vertices generated: 7
Path length: 4.99254 m
Path duration: 9.98295 s
…(not related)
[ WARN] [1720259756.916861766, 630.876000000]: New plan published, stamp = 630.876000
local_planner_node: /usr/include/eigen3/Eigen/src/Core/Block.h:120: Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::Block(XprType&, Eigen::Index) [with XprType = const Eigen::Matrix<double, -1, -1>; int BlockRows = 1; int BlockCols = -1; bool InnerPanel = false; Eigen::Index = long int]: Assertion `(i>=0) && ( ((BlockRows==1) && (BlockCols==XprType::ColsAtCompileTime) && i<xpr.rows()) ||((BlockRows==XprType::RowsAtCompileTime) && (BlockCols==1) && i<xpr.cols()))' failed.
[ INFO] [1720259756.970799916, 630.925000000]: GBP in refine mode
[robot_1/local_planner-2] process has died [pid 31857, exit code -6, cmd /home/ryanz/quad_sdk/devel/lib/local_planner/local_planner_node __name:=local_planner __log:=/home/ryanz/.ros/log/69c7e27a-3b7c-11ef-a341-713571f0e562/robot_1-local_planner-2.log].
log file: /home/ryanz/.ros/log/69c7e27a-3b7c-11ef-a341-713571f0e562/robot_1-local_planner-2*.log
[ERROR] [1720259757.643786429, 631.566000000]: ID node couldn't find the correct ref state!
t_now 0.693, plan front 0, plan back 0.691
[robot_1/body_force_estimator-3] process has died [pid 31858, exit code -11, cmd /home/ryanz/quad_sdk/devel/lib/body_force_estimator/body_force_estimator_node __name:=body_force_estimator __log:=/home/ryanz/.ros/log/69c7e27a-3b7c-11ef-a341-713571f0e562/robot_1-body_force_estimator-3.log].
log file: /home/ryanz/.ros/log/69c7e27a-3b7c-11ef-a341-713571f0e562/robot_1-body_force_estimator-3*.log
[ INFO] [1720259759.040186422, 632.925000000]: GBP in refine mode
```
Any ideas? Thanks a lot!

ryanz2023, 2024-07-06:
Fixed: I built it again with `catkin clean` and `catkin build` and it was solved. 
Still investigating the errors. Helps are welcome.

ryanz2023, 2024-07-07:
This error appears again. Is there any description of error code so I can understand the errors clearly? Thx

ryanz2023, 2024-07-08:
@goldbattle @justinyim @astutt @nikolaif399 Still maintained?

ryanz2023, 2024-07-21:
@ologandavid

ologandavid, 2024-08-06:
Hi @ryanz2023, 
Are you publishing a clicked point at the start of the rviz simulation? The local planner seems to be running too slow are you running the right version of the solver and/or are you feeding a viable goal point to the robot?

Best, 
David Ologan

ryanz2023, 2024-08-14:
I think so. I've checked the log and there's no error about the solver. Probably this is a bug of rviz.

ryanz2023, 2025-07-10:
> Is this still an issue, otherwise I'm going to mark this as closed.

Just close it.

closed 2025-07-10 by ologandavid

wulidede, 2025-11-05:
@ryanz2023 How did you solve this bug? I encountered the same bug.