robomechanics/quad-sdk #428: NMPC solving fail

RoCos206, 2024-08-06:
I solved loading the robot and standing it up, but when I try to run twist, the nmpc solving fail problem occurs.

[ WARN] [1722911802.924340090, 12.319000000]: NMPC solving fail

ologandavid, 2024-08-06:
Hi @RoCos206, 
Can you provide a little more information on exactly what you're doing? What commands are you running, the branch name, solver version you're using etc. 

Best, 
David Ologan

RoCos206, 2024-08-07:
I'm using noetic devel, and while loading the robot I get nmpc soiving fail when I use command roslaunchquad_utilsquad_plan.launch reference:=twistlogging:=true. How can I check the solver version?

RoCos206, 2024-08-07:
Solved! It was a hsl problem

ologandavid, 2024-08-07:
Happy to hear it!

closed 2024-08-07 by ologandavid