robomechanics/quad-sdk #429: A1 fail to run local planner

zhxiaoyu218, 2024-08-20:
I was trying to run play with A1.  But, I was not able to run 
```bash
roslaunch quad_utils quad_plan.launch reference:=twist logging:=true"
```
the error is something like 
```bash
ROS_MASTER_URI=http://localhost:11311

process[robot_1/local_planner-1]: started with pid [4280]
process[robot_1/body_force_estimator-2]: started with pid [4281]
process[robot_1/rosbag_record-3]: started with pid [4282]
process[robot_1/rosbag_record_archive-4]: started with pid [4283]
process[robot_1/plan_publisher-5]: started with pid [4284]
[robot_1/local_planner-1] process has died [pid 4280, exit code -11, cmd /starbox/catkin_ws/devel/lib/local_planner/local_planner_node __name:=local_planner __log:=/root/.ros/log/5a4f2f0c-5f02-11ef-9c62-2c335892e181/robot_1-local_planner-1.log].
log file: /root/.ros/log/5a4f2f0c-5f02-11ef-9c62-2c335892e181/robot_1-local_planner-1*.log
^C[robot_1/plan_publisher-5] killing on exit
[robot_1/rosbag_record_archive-4] killing on exit
[robot_1/rosbag_record-3] killing on exit
[robot_1/body_force_estimator-2] killing on exit
^C^Cshutting down processing monitor...
... shutting down processing monitor complete
```
**I am able to run both local planner and global planner for spirit robot**

ologandavid, 2024-08-23:
What branch are you currently operating on and have you made any changes to the code? I verified that this is working on devel as of today. It's highly likely that this is an issue with your install of the solver.

Best, 
David Ologan

zhxiaoyu218, 2024-08-24:
> What branch are you currently operating on and have you made any changes to the code? I verified that this is working on devel as of today. It's highly likely that this is an issue with your install of the solver.
> 
> Best, David Ologan

Hi, david, thank you so much for your reply! I am using devel branch. if I am able to run spirit robot, is it still likely that there is a issue with solver?
And,  just want to make sure the robot driver of A1 is not required in order to stand up in simulation?