robomechanics/quad-sdk #432: ROS2 Humble Support for Quad SDK

abhishekvchaudhari, 2024-09-08:
I am currently working with the Quad SDK built for ROS1. As I am planning to migrate my project to ROS2, I would like to know if there are any plans or updates to support ROS2, specifically the Humble Hawksbill distribution.

Could you kindly clarify whether this SDK is compatible with ROS2 Humble, or if there are any known efforts or roadmaps in place to update the SDK for ROS2 compatibility? If not, could you suggest any potential migration paths or known challenges in adapting the current ROS1-based implementation for ROS2?

Looking forward to your response.

Thank you...!!

ologandavid, 2024-09-10:
@abhishekvchaudhari,
This is currently a work in progress, I've been working on porting over the packages so the codebase is available in ROS2. Currently, I've been working in Jazzy, but I'm assuming it shouldn't be too different from Humble. I'll try and keep you updated on my progress.

Best, 
David

abhishekvchaudhari, 2024-09-14:
> @abhishekvchaudhari, This is currently a work in progress, I've been working on porting over the packages so the codebase is available in ROS2. Currently, I've been working in Jazzy, but I'm [...]

Thank you, David, for getting back to me.

I am currently working on the quadruped robot development using the quad-sdk in the ROS1 branch. I have a few questions regarding hardware integration and would appreciate your guidance:

Position-Controlled BLDC Motor Integration: How can I integrate position-controlled BLDC motors with the quad-sdk robot driver? Specifically, are there any interfaces or guidelines for handling position, KP, KD, and feedforward terms (ff)?

IMU Sensor Integration: Could you provide some guidance on integrating an IMU sensor with the quad-sdk? Is there a recommended sensor model or package that works best with the current implementation?

Custom Robot URDF Model: Are there any guidelines or best practices for creating a custom URDF model for a quadruped robot that would be compatible with the quad-sdk?

I would greatly appreciate any insights or resources that could assist in integrating the hardware with the quad-sdk.

Looking forward to your feedback.

Best regards,
Abhishek

ologandavid, 2024-09-16:
Hi @abhishekvchaudhari, 
Take a look at the wiki associated with Quad-SDK, I think the sensor integration and URDF questions could be resolved by referencing these sections. https://github.com/robomechanics/quad-sdk/wiki . It might also be helpful to reference the way we implemented our IMU in the robot_driver package.

With regards to the motor gains, these are largely dependent on your system, and we tuned them by hand to land on the desired values we currently use on our hardware platform. 

Let me know if this helps and if you have any additional questions. 

Best, 
David Ologan

ologandavid, 2024-09-30:
@abhishekvchaudhari, Checking in if this works for you, otherwise I'm going to go ahead and close this.

abhishekvchaudhari, 2024-10-01:
> @abhishekvchaudhari, Checking in if this works for you, otherwise I'm going to go ahead and close this.

Hi David, 
  Yes, I am referring to the quad-sdk wiki for hardware integration. You can close the issue.
  Thank you for your support.