robomechanics/quad-sdk #435: robot cannot meet desired height

zhxiaoyu218, 2025-01-09:
We are conducting some experiments, and, however in the global planner, the desired height for Spirit is set to **0.27**. However, the results show that the actual height consistently below the **0.27** and remains around **0.25** in gazebo. is there any parameter I can change?

ologandavid, 2025-01-09:
What branch are you currently working on? On my system, 0.27 m should be the nominal stand body height in gazebo. It was set this way to accurately reflect the body height in real life on hardware. The issue might stem from any modifications you made in the global body planner. See image attached below.
[image]

zhxiaoyu218, 2025-01-10:
Hi, david, thank you so much for your reply. I am using devel branch which is noetic. 

_here is the modification I have made:_
in **local_planner.yaml** the desired height currently is 0.27m.
in **spirit.yaml**, I just modified h_nom from **0.3** to 0.27.

in stand mode, without any walking, it is indeed around 0.27, but around 0.25 while **walking**.

_another question is about hip_offset in matlab for spirit._
`parameter.physics.hip_offset=[0.2263; 0.098; 0];` in matlab code
but it seems it should be [0.2263; 0.07; 0] from spirit.sdf?

closed 2025-03-22 by zhxiaoyu218