robomechanics/quad-sdk #436: Quad-SDK with ROS2

torstenrichteressen, 2025-01-17:
Hi,

is there a quad-sdk Version for ROS2 available?

Thanks
Torsten

ologandavid, 2025-01-17:
Working on this right now and the base code is updated, but I'm working through the auxiliary packages. I'll keep you posted on my progress. 

Best, 
David Ologan

torstenrichteressen, 2025-01-18:
Thanks, It would be great if you kept me up to date. Which ROS2 version are you using?

Best
Torsten

ologandavid, 2025-01-22:
ROS2 Humble since certain packages haven't been upgraded to work with Jazzy.  I'll keep you up to speed as soon as I get all the packages working. 

Best, 
David Ologan

Cdfghglz, 2025-05-02:
Hi @ologandavid , could you please share the progress on this issue?

ologandavid, 2025-07-09:
See devel_ros2 branch. Its still development code, and documentation hasn't been fully written for it, but I'll provide updates there as things progress.