robomechanics/quad-sdk #439: the branch for EKF for the onboard state estimation with mocap

zhxiaoyu218, 2025-03-22:
I want to run the robot without mocap. it noticed on 10/01/2023, the ekf is released which uses imu and kinematic. 
I would to know which branch contains this EKF

jbzhang23, 2025-03-25:
You can use the noetic_devel_ekf_shuo_hardware for noetic for now. This branch works in sim, but will need some editting to work on hardware.
However, the lab is reconstructing due to some issues. In a month or later, there will be some some big updates. The code will be migrating to ros 2 and will support Unitree Go2 as its primary platform.

zhxiaoyu218, 2025-03-27:
Hi, zhang，
thanks for your response, will the update with merging of on board state estimation?

jbzhang23, 2025-04-03:
Yes, I believe so.